pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 #3: Bltouch wiring

codedmind, 2020-02-01:
Hello, can you explain the bltouch wiring... i'm using until now the servos and the z-index to using the bltouch but you refer to use servo and probe?

I'm using this https://camo.githubusercontent.com/56b380ce77969d6bf7e4d627094676b2625dc876/68747470733a2f2f692e696d6775722e636f6d2f42524a516e65392e6a7067

pmjdebruijn, 2020-02-01:
I'm not sure what to explain. My README is fairly clear. Just connect the BLTouch to the PROBE header as opposed to the Z-STOP. SERVO remains unchanged, same in both cases.

pmjdebruijn, 2020-02-01:
If you have a clear defined question, I'd be happy to answer though...

codedmind, 2020-02-01:
@pmjdebruijn sorry if i'm not able to explain... the wiring you say in the readme is like the picture i share?

when you right the z header isn't used is in the board or in the machine?

pmjdebruijn, 2020-02-01:
My README explicitly states the Z-STOP is ignored. So I'm not sure why you're confused about this.

Directly next to the SERVO header, there is a header clearly labeled as PROBE, as is referred to by my README.

As for the exact wiring, that highly depend on where you got your BLtouch from, cross-reference the SKR mini E3 board documentation with the pin labels on your BLtouch.

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2PIN.pdf

codedmind, 2020-02-01:
@pmjdebruijn  ok thanks for clarify that. Like i said i see that wiring scheme that is the same as the teaching tech if i'm not wrong, and used it that way (using the z-stop) so apparently the bltouch could be wiring in two different ways :/

pmjdebruijn, 2020-02-01:
Indeed.

The PROBE header is specifically intended for the BLTouch, but all boards have this luxury.

Connecting the BLTouch to the Z-STOP is essentially a workaround for cost-reduced boards (like the Creality Melzi) that do not have a dedicated connector for it.

I've left the original Z-STOP micro switch connected, so I can flash back to a non-bltouch firmware at any point, and the Z-STOP micro switch will just work again as normal, without having to do any rewiring.

closed 2020-02-01 by pmjdebruijn

datarager, 2020-03-30:
ik heb deze https://www.amazon.de/gp/product/B07Y4DXNNN/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1
maar de draden zijn anders van kleur is origineel zeg maar moet ik hier ook de draden omwisselen

pmjdebruijn, 2020-03-30:
As I've already mentioned above, you should cross reference the markings on your BLTouch PCB with the documentation of your specific board.