pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 #6: Osx script

codedmind, 2020-03-14:
Hello,

Trying to use the script in OSX to compile the firmware to try linear advance and getting sed errors.

sed: 1: "Marlin/platformio.ini": invalid command code M

codedmind, 2020-03-14:
Changing all the sed lines with '' after the -i solve that issue, but raise another

Error: Unknown environment names 'STM32F103RC_bigtree_512K'. Valid names are 'STM32F103RE_btt, STM32F103RE_btt_USB, rambo, STM32F103RC_btt_512K_USB, ARMED, sanguino644p, mks_robin_lite3, STM32F103RC, at90usb1286_cdc, mks_robin, STM32F407VE_black, mega2560, STM32F103RC_fysetc, sanguino1284p, STM32F103RC_btt_USB, mega1280, LPC1769, STM32F401VE_STEVAL, melzi, chitu_f103, rumba32_mks, DUE_USB, STM32F7, teensy31, teensy35, STM32F4, mks_robin_pro, STM32F103VE_GTM32, FLYF407ZG, STM32F103RE, STM32F103RC_btt_512K, at90usb1286_dfu, BIGTREE_BTT002, SAMD51_grandcentral_m4, STM32F103CB_malyan, include_tree, mks_robin_lite, BIGTREE_GTR_V1_0, FYSETC_F6_14, jgaurora_a5s_a1, LPC1768, FYSETC_F6_13, mks_robin_mini, linux_native, esp32, STM32F103RC_btt, BIGTREE_SKR_PRO, rumba32_f446ve, melzi_optiboot, mks_robin_nano, DUE, DUE_debug, FYSETC_S6, STM32F103VE_longer'

codedmind, 2020-03-14:
Ok changed to STM32F103RC_btt_512K and everything works

closed 2020-03-14 by codedmind

pmjdebruijn, 2020-03-14:
Jup, seems to be an upstream change, I'll update my script over the next week or so.

codedmind, 2020-03-14:
After upgrade the firmware it appears the printer only use 1/4 of the bed
compared to a.previous firmware (the one the you provide compiled) :/

pmjdebruijn, 2020-03-14:
See either the README or #5 (which I've pinned to remain visible for exactly this reason)

pmjdebruijn, 2020-03-16:
Also new build script available which fixes this particular issue.