pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 #7: Strange sound while Bed levelling

DaLukasTI, 2020-03-21:
Just wanted to check if the "strange" sounds I hear when the x/y axis is moving while leveling the bed (G29) is normal and ok, sounds a little like "Bvvvvt".
It only makes these sounds in bed levelling not while printing/homing.
Haven't heard these sounds on other firmwares.

pmjdebruijn, 2020-03-21:
I can only say that I do hear a slightly different sound hole bedlevelling as well, but that might just be due to the different movement speeds. I've never really looked into it.

DaLukasTI, 2020-03-21:
Was thinking the same, that's probably fine. Anyway I have another question with the z axis offset. It seems I can't change my z-offset whilst I'm printing (didn't try babystepping just the z-offset). It didn't lower the nozzle. I have to lower the Z offset then home and then level corners to see the effect of the offset being changed. Is this normal behaviour?

pmjdebruijn, 2020-03-21:
No clue to be honest. As far as I'm aware that's not specific to my firmware though.

The only thing to be aware of is that I do have enabled the BABYSTEPPING multiplier to 5.

closed 2020-03-24 by DaLukasTI

pmjdebruijn, 2020-03-25:
I think I know what's going on here, in recent versions I lowered the X and Y HYBRID THRESHOLD:
#define X_HYBRID_THRESHOLD     120
#define Y_HYBRID_THRESHOLD     120

With the HOMING_FEEDRATE being equal or more:
#define HOMING_FEEDRATE_XY (20*60)

During homing, when the movement reaches top speed, the TMC stepper switches to the louder spreadcycle as oppossed to stealthchop.

I'll raise those defaults again, in my next firmware build. 

In the meantime you can raise the TMC Hybrid Threshold via the Configuration menu.