pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 #8: printing randomly out of bed

DaLukasTI, 2020-03-24:
Another weird issue I've encountered is that sometimes when i start a print via octoprint, he starts the print fine heating up the bed and extruder etc.
But after the levelling part instead of going to the point he should be printing he starts printing mid air after the 9th probe. if I reset the printer and rerun the same file again from octoprint he prints perfectly fine.
This happend 3 times at random occasions. Any clue?

pmjdebruijn, 2020-03-24:
No sorry, I've never used Octoprint... 

But if anything that's likely an upstream Marlin issue, or an upstream Octoprint issue....

BTW, are you using my start/stop gcodes as well?

pmjdebruijn, 2020-03-24:
BTW, how high does it print "in the air"?

DaLukasTI, 2020-03-26:
Was about to try your start and stop gcodes in my next print I slice. The height you get when probing the last point and then raising the z-axis as it does with all the probes. But instead of lowering and going to the x/y origin and drawing the priming line to clear the nozzle it just stays at that +-10 z and starts "printing". Yet it doesn't move it's head either it just spills out filament as it were printing.

pmjdebruijn, 2020-03-28:
Did changing the start/end gcode change anything?

DaLukasTI, 2020-03-29:
Just updated to latest commit and making new print with your start/stop gcode.
I'll keep u posted.

DaLukasTI, 2020-04-03:
Closing this issue, after a few prints I can confirm that the start and stop gcode does a good job fixing these issue. Also this startcode is a lot better then the default cure startcode. Did switch over to prusaslicer which seems to be a lot better aswell!

closed 2020-04-03 by DaLukasTI