pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 #9: Full files available

annoyingmosquito, 2020-03-29:
Is it possible to download the full file version of you build?
Want to test some settings regarding the bed probing

annoyingmosquito, 2020-03-29:
I want to test AUTO_BED_LEVELING_BILINEAR for a grid of 5x5 instead of the bed leveling that is now 3x3.
Just to see if printing will be easier on a 'bended bed'

DaLukasTI, 2020-03-29:
You could edit the .sh file at line 226 there is:
`sed -i 's@.*#define GRID_MAX_POINTS_X .*@  #define GRID_MAX_POINTS_X 3@' ${MARLIN_DIR}/Marlin/Configuration.h`
Change this to the amount of probe points you'd like. eg:
`sed -i 's@.*#define GRID_MAX_POINTS_X .*@  #define GRID_MAX_POINTS_X 5@' ${MARLIN_DIR}/Marlin/Configuration.h`
This will result in a 5x5 grid.
Run the .sh again and it'll compile the new firmware.

pmjdebruijn, 2020-03-30:
@DaLukasTI thanks!

closed 2020-03-30 by pmjdebruijn