pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 #14: can you make a copy but compatible with the older bltouches

WalterWilson337, 2020-07-18:
thx

pmjdebruijn, 2020-07-19:
I just publish what I use myself. So others can benefit from my work if it suits them.

If you need something different you're free to modify my script to suit your own needs, and build your own firmware.

WalterWilson337, 2020-07-20:
can you change the probing speed down I don't know what you changed and you don't have the un compiled

WalterWilson337, 2020-07-20:
I can't make my one I have tried and I can't downgrade to my former firmware because it's much too outdated

pmjdebruijn, 2020-07-20:
> can you change the probing speed down I don't know what you changed and you don't have the un compiled

Again, the firmware does whatever I want it to do. I don't customize it for other people. If you want something different, that's something you'll need to work out yourself.

Given this is a git repository, you can trivially see, what I changed at any point in time, in the commit history:

https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2/commits/master

Specifically this:
https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2/commit/e0581bee3d079bd101b3f9ff956d012bbee333c3#diff-34a97950ad5ef517e78521c4a8d6799fR231

My build script skr_mini_e3_build.sh basically does everything it needs for build a firmware, including pulling in the relevant dependencies. It does however need to be run on a Linux host, but that's easy enough to install on a virtual machine if you're running Windows.

> I can't make my one I have tried and I can't downgrade to my former firmware because it's much too outdated

There a huge amount of information available online how to customize your own firmware, so given enough effort anybody can.

But, there always the official reference firmware from BigTreeTech themselves:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2  (I never tried those myself though)

WalterWilson337, 2020-07-20:
if you could do it for me I can pay

WalterWilson337, 2020-07-21:
if any one makes the same config but but compatible with an older bltouch

WalterWilson337, 2020-07-21:
I can pay

pmjdebruijn, 2020-07-21:
Sorry, I'm not looking to take on commission work.

Maybe you can find a local makerspace, where someone might be able to help you out.

closed 2020-07-21 by pmjdebruijn