pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 #15: Can't compile firmware on ubuntu but does compile on arch even though .bin is corrupt

DaLukasTI, 2020-09-27:
I tried to use your latest script and the firmware.bin it produced on my arch linux machine doesn't work.
When I flash the generated .bin a lot of options are missing in the menu's and I don't have options to heat the bed or see the temperature of the bed on the main screen, homing is not working etc... . Anyway when flashing the firmware.bin from your repo does work perfectly. The bin produced by the shell script is 30kb smaller than yours even if let it unedited and just compile it which seems odd.

When I tried to compile the firmware on ubuntu (tried kubuntu 18.04 & 20.04.4 LTS) it couldn't install Platformio with pip throwing a dist_wheel error. also the "==" operator on line 18 & 21 of the script don't get recognized in ubuntu but does work just fine in arch and manjaro.

I never had issues using your shell script and I was wondering if u had any ideas because I can't get it to work.

Thanks in advance

pmjdebruijn, 2020-09-29:
One issue on my current script is that the Marlin codebase tracks a branch which is dynamic, and the Configurations track a tag which is static, so it always breaks after some time due to upstream Marlin changes. Just change the 2.0.x branch to 2.0.6.1 release, and it'll probably work just fine again.

I also noticed 2.0.7 being released, so I'll probably update everything somewhere during next weekend or so.

DaLukasTI, 2020-09-29:
I'll try changing the branch and if it doesn't work I'll stick with your prebuild firmware.bin and wait for you to update to 2.0.7.

DaLukasTI, 2020-09-29:
Changing the branch still results in a .bin which is still 30 ish kb  smaller than yours, so my guess is it still won't work if I flash it (it's only 0.4 kb bigger then the previous one.) So I'll keep using your prebuild .bin until you release an update for 2.0.7 :).

closed 2020-09-29 by DaLukasTI

pmjdebruijn, 2020-10-10:
Just pushed 2.0.7.1