pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 #16: Can't build wheel for platformio

DaLukasTI, 2020-10-10:
When I run the bash script I get an error when building wheel for platformio.
[
[image]
](url)
And I can't find the problem.
Thanks in advance

pmjdebruijn, 2020-10-11:
Why do you presume it matters? That's just some python nonsense, the script should proceed to build a firmware.bin just fine.

If you're still not getting a proper firmware.bin, please post a full  build log, in proper text, and not a screenshot of a terminal with a particularly poor color scheme.

pmjdebruijn, 2020-10-19:
This particular error message has been resolved in the latest version I just pushed.

pmjdebruijn, 2020-10-24:
Given the reported issue was resolved, and lacking any new feedback, I'll be closing this issue.

closed 2020-10-24 by pmjdebruijn